pimoroni/unicorn-hat #18: cannot get it work on Raspberry Pi 2 Model B

curcuz, 2015-02-09:
I followed the documentation as usal and installed the python/pip dependencies, then installed unicornhat via pip. Every attempt on running any of the examples ends up with a clean script execution with no errors, but the hat actually does nothing. 

here are the output of the pip installation of unicornhat:
```
Downloading/unpacking unicornhat
  Downloading unicornhat-1.0.2.tar.gz
  Running setup.py egg_info for package unicornhat
    
Downloading/unpacking ws2812 (from unicornhat)
  Downloading ws2812-1.0.0.tar.gz (58Kb): 58Kb downloaded
  Running setup.py egg_info for package ws2812
    
    warning: no files found matching 'ws2812-RPi.h'
    warning: no files found matching 'ws2812-RPi.c'
    warning: no files found matching 'LICENSE.txt'
Installing collected packages: unicornhat, ws2812
  Running setup.py install for unicornhat
    
  Running setup.py install for ws2812
    building '_ws2812' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.7 -c lib/ws2812-RPi.c -o build/temp.linux-armv7l-2.7/lib/ws2812-RPi.o
    In file included from lib/ws2812-RPi.c:1:0:
    lib/ws2812-RPi.h:95:1: warning: multi-line comment [-Wcomment]
    In file included from lib/ws2812-RPi.c:1:0:
    lib/ws2812-RPi.h:123:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:389:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:436:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:456:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:470:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:481:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.c: In function 'reverseWord':
    lib/ws2812-RPi.c:49:16: warning: variable 'bit' set but not used [-Wunused-but-set-variable]
    lib/ws2812-RPi.c: At top level:
    lib/ws2812-RPi.c:147:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.c:292:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.c:471:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.c: In function 'initHardware':
    lib/ws2812-RPi.c:569:3: warning: suggest parentheses around comparison in operand of '&' [-Wparentheses]
    lib/ws2812-RPi.c: At top level:
    lib/ws2812-RPi.c:745:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.c: In function 'show':
    lib/ws2812-RPi.c:829:1: warning: "/*" within comment [-Wcomment]
    lib/ws2812-RPi.c:762:10: warning: unused variable 'color' [-Wunused-variable]
    lib/ws2812-RPi.c:758:15: warning: unused variable 'PWMWaveformBitPos' [-Wunused-variable]
    lib/ws2812-RPi.c:757:15: warning: unused variable 'LEDBuffeWordPos' [-Wunused-variable]
    lib/ws2812-RPi.c: At top level:
    lib/ws2812-RPi.c:839:1: warning: multi-line comment [-Wcomment]
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.7 -c ws2812-RPi_wrap.c -o build/temp.linux-armv7l-2.7/ws2812-RPi_wrap.o
    In file included from ws2812-RPi_wrap.c:2967:0:
    lib/ws2812-RPi.h:95:1: warning: multi-line comment [-Wcomment]
    In file included from ws2812-RPi_wrap.c:2967:0:
    lib/ws2812-RPi.h:123:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:389:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:436:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:456:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:470:1: warning: multi-line comment [-Wcomment]
    lib/ws2812-RPi.h:481:1: warning: multi-line comment [-Wcomment]
    ws2812-RPi_wrap.c: In function 'init_ws2812':
    ws2812-RPi_wrap.c:4722:21: warning: variable 'md' set but not used [-Wunused-but-set-variable]
    ws2812-RPi_wrap.c: At top level:
    lib/ws2812-RPi.h:336:17: warning: 'virtbase' defined but not used [-Wunused-variable]
    lib/ws2812-RPi.h:338:31: warning: 'pwm_reg' defined but not used [-Wunused-variable]
    lib/ws2812-RPi.h:339:31: warning: 'clk_reg' defined but not used [-Wunused-variable]
    lib/ws2812-RPi.h:340:31: warning: 'dma_reg' defined but not used [-Wunused-variable]
    lib/ws2812-RPi.h:341:31: warning: 'gpio_reg' defined but not used [-Wunused-variable]
    lib/ws2812-RPi.h:359:31: warning: 'ctl' defined but not used [-Wunused-variable]
    gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro build/temp.linux-armv7l-2.7/lib/ws2812-RPi.o build/temp.linux-armv7l-2.7/ws2812-RPi_wrap.o -Llib/ -o build/lib.linux-armv7l-2.7/_ws2812.so
    
    warning: no files found matching 'ws2812-RPi.h'
    warning: no files found matching 'ws2812-RPi.c'
    warning: no files found matching 'LICENSE.txt'
Successfully installed unicornhat ws2812
Cleaning up...
```

then i downloaded an example from this repo and tried to run that:
```
pi@fwrpi ~ $ wget https://raw.githubusercontent.com/pimoroni/UnicornHat/master/python/examples/matrix.py
--2015-02-09 17:54:50--  https://raw.githubusercontent.com/pimoroni/UnicornHat/master/python/examples/matrix.py
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.31.17.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.31.17.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 981 [text/plain]
Saving to: `matrix.py.1'

100%[===========================================================================================>] 981         --.-K/s   in 0.001s  

2015-02-09 17:54:54 (1.28 MB/s) - `matrix.py.1' saved [981/981]

pi@fwrpi ~ $ chmod +x matrix.py
pi@fwrpi ~ $ sudo python matrix.py
^CTraceback (most recent call last):
  File "matrix.py", line 23, in <module>
    unicorn.show()
  File "/usr/local/lib/python2.7/dist-packages/unicornhat.py", line 117, in show
    ws2812.show()
KeyboardInterrupt
pi@fwrpi ~ $ 

```

waited several seconds (up to 30) before that ctrl+c with no effect on the hat...

Gadgetoid, 2015-02-09:
This is something to do with the Pi 2, and we're working on it. You can keep up-to-date on what does and doesn't work on the Pi 2 here: http://forums.pimoroni.com/t/raspberry-pi-2-what-works-and-what-doesnt/352/2

curcuz, 2015-02-09:
thanks for the clarification, i'll follow that thread!

closed 2015-03-14 by Gadgetoid