pimoroni/unicorn-hat #22: unicornd daemon install fails

ukmoose, 2015-03-14:
If I try to install the daemon from ~/unicorn-hat/c/unicornd
using sudo make install

I get the following error message:

cp unicornd /usr/sbin
cp: cannot stat `unicornd': No such file or directory
Makefile:11: recipe for target 'install' failed
make: *** [install] Error 1

Gadgetoid, 2015-03-14:
Whoops! This is because of an incredibly stupid thing I did in https://github.com/pimoroni/unicorn-hat/commit/656556fd817d6ea6f1d1e27f5371b23646658953 and then haven't had time to fix since. 

It's now back in https://github.com/pimoroni/unicorn-hat/commit/cabf748a9d36fa28a46c14af2945a3e39941a44f for your git-cloning and installing pleasure!

Gadgetoid, 2015-03-14:
Actually it might be due to the makefile not inheriting "make" when running "install". Try:

```bash
sudo make
sudo make install
```

Gadgetoid, 2015-03-14:
Fixed the Makefile ( somewhat ) and documentation: https://github.com/pimoroni/unicorn-hat/commit/2235e1b37666396909e9c6a759a88c6044c5d8bb

Process is now:

```bash
sudo make
sudo make install
sudo service unicorn start
```

And verify that it works with:

```bash
cd test_client
perl unicorn_demo.pl
```

Gadgetoid, 2015-03-15:
And for when you get it up and running, and if Python is your thang, have an example Python script and library for using Unicornd: https://github.com/pimoroni/unicorn-hat/commit/0bcd02992d9a1550e4fad2f296e8b1ebd6a40e7b

ukmoose, 2015-03-16:
Many thanks for the prompt reply!  I can confirm it now works.

closed 2015-03-16 by ukmoose