pimoroni/unicorn-hat #27: Testing with RPi 2?

wannabeCitizen, 2015-04-10:
I am still seeing lingering issues of errors with RPi 2 and can confirm that running today, trying to control a 50 LED strand of WS2812's, python would hang when trying to create the channel for the strip.

Has there been any luck that I don't know of getting it to work with Pi 2?  Anything that's been identified as a culprit to this problem?

Gadgetoid, 2015-04-10:
As far as I'm aware the Pi 2 compatibility is done and dusted- it should absolutely work, that's to say we've had it running non-stop for several days! Which library are you trying to use?

wannabeCitizen, 2015-04-10:
I was using the python rpi_ws281x library.  But that's good to hear it's working.  Were you running one of the python bindings?  If so, which one?  I'll try it out tonight.

closed 2016-03-21 by Gadgetoid