pimoroni/unicorn-hat #31: pip[3] install fails for unicornhat

t3chguy, 2015-08-04:
See log below:
```
t3chguy@Admin-RPi ~> sudo pip3 install unicornhat
Downloading/unpacking unicornhat
  Downloading unicornhat-2.0.4.tar.gz
  Running setup.py (path:/tmp/pip-build-1yzbpsjh/unicornhat/setup.py) egg_info for package unicornhat

Downloading/unpacking rpi-ws281x>=1.1.0 (from unicornhat)
  Downloading rpi_ws281x-1.1.3.tar.gz (102kB): 102kB downloaded
  Running setup.py (path:/tmp/pip-build-1yzbpsjh/rpi-ws281x/setup.py) egg_info for package rpi-ws281x

Installing collected packages: unicornhat, rpi-ws281x
  Running setup.py install for unicornhat

  Running setup.py install for rpi-ws281x
    Compiling ws281x library...
    building '_rpi_ws281x' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-stron                                       g -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -Ilib/ -I/usr/include/python3.4m -c rpi_ws281x_wrap.                                       c -o build/temp.linux-armv7l-3.4/rpi_ws281x_wrap.o
    rpi_ws281x_wrap.c: In function ‘PyInit__rpi_ws281x’:
    rpi_ws281x_wrap.c:4638:21: warning: variable ‘md’ set but not used [-Wunused-but-set-variable]
       PyObject *m, *d, *md;
                         ^
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,rel                                       ro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_                                       SOURCE=2 build/temp.linux-armv7l-3.4/rpi_ws281x_wrap.o -Llib/ -lws2811 -o build/lib.linux-armv7l-3.4/_rpi_ws281x.                                       cpython-34m.so
    /usr/bin/ld: lib//libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be                                        used when making a shared object; recompile with -fPIC
    lib//libws2811.a: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-1yzbps                                       jh/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __fi                                       le__, 'exec'))" install --record /tmp/pip-3z7pd__f-record/install-record.txt --single-version-externally-managed                                        --compile:
    running install

running build

running build_py

Compiling ws281x library...

creating build

creating build/lib.linux-armv7l-3.4

copying neopixel.py -> build/lib.linux-armv7l-3.4

running build_ext

building '_rpi_ws281x' extension

creating build/temp.linux-armv7l-3.4

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -W                                       format -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -Ilib/ -I/usr/include/python3.4m -c rpi_ws281x_wrap.c -o                                        build/temp.linux-armv7l-3.4/rpi_ws281x_wrap.o

rpi_ws281x_wrap.c: In function ‘PyInit__rpi_ws281x’:

rpi_ws281x_wrap.c:4638:21: warning: variable ‘md’ set but not used [-Wunused-but-set-variable]

   PyObject *m, *d, *md;

                     ^

arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -                                       Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOUR                                       CE=2 build/temp.linux-armv7l-3.4/rpi_ws281x_wrap.o -Llib/ -lws2811 -o build/lib.linux-armv7l-3.4/_rpi_ws281x.cpyt                                       hon-34m.so

/usr/bin/ld: lib//libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be use                                       d when making a shared object; recompile with -fPIC

lib//libws2811.a: error adding symbols: Bad value

collect2: error: ld returned 1 exit status

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/tmp/pip-build-1yzbpsjh/rpi-ws281x/setup.py';e                                       xec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install -                                       -record /tmp/pip-3z7pd__f-record/install-record.txt --single-version-externally-managed --compile failed with err                                       or code 1 in /tmp/pip-build-1yzbpsjh/rpi-ws281x
Storing debug log for failure in /home/t3chguy/.pip/pip.log
```

Exactly the same happens using `pip install unicornhat`

Gadgetoid, 2015-08-04:
Could you supply some pertinent details on your OS?

```bash
cat /etc/*-release
ld -v
gcc -v
uname -a
```

t3chguy, 2015-08-04:
I will do shortly, currently trying another image to see whether its directly related to that.
It was the Ubuntu Mate image available on the Raspberry Pi Downloads Page, I will get back here after testing Raspbian, and go back to Mate to get you the details you need.

Thanks

Gadgetoid, 2015-08-04:
We currently don't test on, or support, anything but Raspbian and I've got a suspicion that Unicorn HAT will be particularly tricky to get working with Ubuntu Mate due to the somewhat arcane way in which it works.

I'll never say never, though. I'm downloading Ubuntu Mate right now to see if I can get it working myself. I'm currently supposed to be busy with something else, though, so I might be a while getting back to you!

t3chguy, 2015-08-04:
Thanks for testing it yourself regardless! Raspbian is almost installed on a secondary card so I can test if it works for me there.

t3chguy, 2015-08-04:
As expected, works blindingly on Raspbian

t3chguy, 2015-08-04:
Although on Raspbian its not working on Py2-6 or -7 for me:
```
pi@raspberrypi ~/quincerly/python/examples $ sudo pip-2.7 install unicornhat
Downloading/unpacking unicornhat
  Downloading unicornhat-2.0.4.tar.gz
  Running setup.py egg_info for package unicornhat

Downloading/unpacking rpi-ws281x>=1.1.0 (from unicornhat)
  Downloading rpi_ws281x-1.1.3.tar.gz (102Kb): 102Kb downloaded
  Running setup.py egg_info for package rpi-ws281x

Installing collected packages: unicornhat, rpi-ws281x
  Running setup.py install for unicornhat

  Running setup.py install for rpi-ws281x
    Compiling ws281x library...
    building '_rpi_ws281x' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o
    rpi_ws281x_wrap.c:125:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools;__file__='/home/pi/quincerly/python/examples/build/rpi-ws281x/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-sK7kYL-record/install-record.txt:
    running install

running build

running build_py

Compiling ws281x library...

creating build

creating build/lib.linux-armv7l-2.7

copying neopixel.py -> build/lib.linux-armv7l-2.7

running build_ext

building '_rpi_ws281x' extension

creating build/temp.linux-armv7l-2.7

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o

rpi_ws281x_wrap.c:125:20: fatal error: Python.h: No such file or directory

compilation terminated.

error: command 'gcc' failed with exit status 1

----------------------------------------
Command /usr/bin/python -c "import setuptools;__file__='/home/pi/quincerly/python/examples/build/rpi-ws281x/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-sK7kYL-record/install-record.txt failed with error code 1 in /home/pi/quincerly/python/examples/build/rpi-ws281x
Storing complete log in /root/.pip/pip.log
pi@raspberrypi ~/quincerly/python/examples $ sudo pip-2. install unicornhat
pip-2.6  pip-2.7
pi@raspberrypi ~/quincerly/python/examples $ sudo pip-2.6 install unicornhat
Downloading/unpacking unicornhat
  Running setup.py egg_info for package unicornhat
Downloading/unpacking rpi-ws281x>=1.1.0 (from unicornhat)
  Running setup.py egg_info for package rpi-ws281x
Installing collected packages: unicornhat, rpi-ws281x
  Running setup.py install for unicornhat
  Running setup.py install for rpi-ws281x
    Compiling ws281x library...
    building '_rpi_ws281x' extension
    gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.6 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.6/rpi_ws281x_wrap.o
    rpi_ws281x_wrap.c:125:20: fatal error: Python.h: No such file or directory
    compilation terminated.
    error: command 'gcc' failed with exit status 1
    Complete output from command /usr/bin/python2.6 -c "import setuptools;__file__='/home/pi/quincerly/python/examples/build/rpi-ws281x/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-Hi67Br-record/install-record.txt:
    running install

running build

running build_py

Compiling ws281x library...

creating build/lib.linux-armv7l-2.6

copying neopixel.py -> build/lib.linux-armv7l-2.6

running build_ext

building '_rpi_ws281x' extension

creating build/temp.linux-armv7l-2.6

gcc -pthread -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fPIC -Ilib/ -I/usr/include/python2.6 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.6/rpi_ws281x_wrap.o

rpi_ws281x_wrap.c:125:20: fatal error: Python.h: No such file or directory

compilation terminated.

error: command 'gcc' failed with exit status 1

----------------------------------------
Command /usr/bin/python2.6 -c "import setuptools;__file__='/home/pi/quincerly/python/examples/build/rpi-ws281x/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --single-version-externally-managed --record /tmp/pip-Hi67Br-record/install-record.txt failed with error code 1 in /home/pi/quincerly/python/examples/build/rpi-ws281x
Storing complete log in /root/.pip/pip.log
```

t3chguy, 2015-08-04:
Trying to install rpi-ws281x on its own also fails:
```
t3chguy@Admin-RPi ~> sudo pip install rpi-ws281x
Downloading/unpacking rpi-ws281x
  Downloading rpi_ws281x-1.1.3.tar.gz (102kB): 102kB downloaded
  Running setup.py (path:/tmp/pip-build-Ao72KX/rpi-ws281x/setup.py) egg_info for package rpi-ws281x

Installing collected packages: rpi-ws281x
  Running setup.py install for rpi-ws281x
    Compiling ws281x library...
    building '_rpi_ws281x' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ilib/ -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o
    rpi_ws281x_wrap.c: In function ‘init_rpi_ws281x’:
    rpi_ws281x_wrap.c:4638:21: warning: variable ‘md’ set but not used [-Wunused-but-set-variable]
       PyObject *m, *d, *md;
                         ^
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o -Llib/ -lws2811 -o build/lib.linux-armv7l-2.7/_rpi_ws281x.so
    /usr/bin/ld: lib//libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
    lib//libws2811.a: error adding symbols: Bad value
    collect2: error: ld returned 1 exit status
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
    Complete output from command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-Ao72KX/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-OHyTLA-record/install-record.txt --single-version-externally-managed --compile:
    running install

running build

running build_py

Compiling ws281x library...

creating build

creating build/lib.linux-armv7l-2.7

copying neopixel.py -> build/lib.linux-armv7l-2.7

running build_ext

building '_rpi_ws281x' extension

creating build/temp.linux-armv7l-2.7

arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -Ilib/ -I/usr/include/python2.7 -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o

rpi_ws281x_wrap.c: In function ‘init_rpi_ws281x’:

rpi_ws281x_wrap.c:4638:21: warning: variable ‘md’ set but not used [-Wunused-but-set-variable]

   PyObject *m, *d, *md;

                     ^

arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -fno-strict-aliasing -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wl,-Bsymbolic-functions -Wl,-z,relro -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security build/temp.linux-armv7l-2.7/rpi_ws281x_wrap.o -Llib/ -lws2811 -o build/lib.linux-armv7l-2.7/_rpi_ws281x.so

/usr/bin/ld: lib//libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC

lib//libws2811.a: error adding symbols: Bad value

collect2: error: ld returned 1 exit status

error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

----------------------------------------
Cleaning up...
Command /usr/bin/python -c "import setuptools, tokenize;__file__='/tmp/pip-build-Ao72KX/rpi-ws281x/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-OHyTLA-record/install-record.txt --single-version-externally-managed --compile failed with error code 1 in /tmp/pip-build-Ao72KX/rpi-ws281x
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    load_entry_point('pip==1.5.6', 'console_scripts', 'pip')()
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 248, in main
    return command.main(cmd_args)
[... 10 lines cut ...]
DISTRIB_CODENAME=vivid
DISTRIB_DESCRIPTION="Ubuntu 15.04"
NAME="Ubuntu"
VERSION="15.04 (Vivid Vervet)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 15.04"
VERSION_ID="15.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
t3chguy@Admin-RPi ~> ld -v
GNU ld (GNU Binutils for Ubuntu) 2.25
t3chguy@Admin-RPi ~> gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 4.9.2-10ubuntu13' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-uniqu [... cut ...] nable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-multilib --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Ubuntu/Linaro 4.9.2-10ubuntu13)
t3chguy@Admin-RPi ~> uname -a
Linux Admin-RPi 3.18.0-25-rpi2 #26-Ubuntu SMP PREEMPT Sun Jul 5 06:46:34 UTC 2015 armv7l armv7l armv7l GNU/Linux

```

Gadgetoid, 2015-08-04:
Managed to get it to compile by just adding `-fPIC` into the library Makefile. I can't push this as a fix until I've tested it on Raspbian, but I've got a working Unicorn HAT on Ubuntu Mate- much to my surprise!

You should replace your `python/rpi-ws281x/lib/Makefile` with:

```
all: lib test

lib:
	gcc -o ws2811.o -c -g -O2 -Wall -Werror ws2811.c -fPIC
	gcc -o pwm.o -c -g -O2 -Wall -Werror pwm.c -fPIC
	gcc -o dma.o -c -g -O2 -Wall -Werror dma.c -fPIC
	gcc -o board_info.o -c -g -O2 -Wall -Werror board_info.c -fPIC
	gcc -o mailbox.o -c -g -O2 -Wall -Werror mailbox.c -fPIC
	ar rc libws2811.a ws2811.o pwm.o dma.o board_info.o mailbox.o
	ranlib libws2811.a

test:
	gcc -o main.o -c -g -O2 -Wall -Werror main.c
	gcc -o test main.o libws2811.a

clean:
	-rm ws2811.o pwm.o dma.o board_info.o mailbox.o
	-rm libws2811.a
	-rm test
```

t3chguy, 2015-08-05:
@Gadgetoid I tried your rendition of the Makefile, still got a fail on installing the UnicornHat py lib:
```
t3chguy@Admin-RPi ~/u/p/UnicornHat> sudo python setup.py install
running install
/usr/lib/python2.7/dist-packages/pkg_resources/__init__.py:2512: PEP440Warning: 'python-apt (0.9.3.11build1)' is being parsed as a legacy, non PEP 440, version. You may find odd behavior and sort order. In particular it will be sorted as less than 0.0. It is recommend to migrate to PEP 440 compatible versions.
  PEP440Warning,
running bdist_egg
running egg_info
creating unicornhat.egg-info
writing requirements to unicornhat.egg-info/requires.txt
writing unicornhat.egg-info/PKG-INFO
writing top-level names to unicornhat.egg-info/top_level.txt
writing dependency_links to unicornhat.egg-info/dependency_links.txt
writing manifest file 'unicornhat.egg-info/SOURCES.txt'
reading manifest file 'unicornhat.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'unicornhat.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
copying unicornhat.py -> build/lib.linux-armv7l-2.7
creating build/bdist.linux-armv7l
creating build/bdist.linux-armv7l/egg
copying build/lib.linux-armv7l-2.7/unicornhat.py -> build/bdist.linux-armv7l/egg
byte-compiling build/bdist.linux-armv7l/egg/unicornhat.py to unicornhat.pyc
creating build/bdist.linux-armv7l/egg/EGG-INFO
copying unicornhat.egg-info/PKG-INFO -> build/bdist.linux-armv7l/egg/EGG-INFO
copying unicornhat.egg-info/SOURCES.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying unicornhat.egg-info/dependency_links.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying unicornhat.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying unicornhat.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
zip_safe flag not set; analyzing archive contents...
creating dist
creating 'dist/unicornhat-2.0.2-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing unicornhat-2.0.2-py2.7.egg
Copying unicornhat-2.0.2-py2.7.egg to /usr/local/lib/python2.7/dist-packages
Adding unicornhat 2.0.2 to easy-install.pth file

Installed /usr/local/lib/python2.7/dist-packages/unicornhat-2.0.2-py2.7.egg
Processing dependencies for unicornhat==2.0.2
Searching for rpi-ws281x>=1.0.1
Reading https://pypi.python.org/simple/rpi_ws281x/
Best match: rpi-ws281x 1.1.3
Downloading https://pypi.python.org/packages/source/r/rpi_ws281x/rpi_ws281x-1.1.3.tar.gz#md5=37284a0fc90d5a82e0faef8c8542c564
Processing rpi_ws281x-1.1.3.tar.gz
Writing /tmp/easy_install-ELYhO3/rpi_ws281x-1.1.3/setup.cfg
Running rpi_ws281x-1.1.3/setup.py -q bdist_egg --dist-dir /tmp/easy_install-ELYhO3/rpi_ws281x-1.1.3/egg-dist-tmp-PhlJaw
Compiling ws281x library...
rpi_ws281x_wrap.c: In function ‘init_rpi_ws281x’:
rpi_ws281x_wrap.c:4638:21: warning: variable ‘md’ set but not used [-Wunused-but-set-variable]
   PyObject *m, *d, *md;
                     ^
/usr/bin/ld: lib//libws2811.a(ws2811.o): relocation R_ARM_THM_MOVW_ABS_NC against `a local symbol' can not be used when making a shared object; recompile with -fPIC
lib//libws2811.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
error: Setup script exited with error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

I really appreciate you getting it to work, hopefully it'll also work on Raspbian so this change can be pushed into the repo.

Thanks

t3chguy, 2015-08-05:
@Gadgetoid I can confirm that even with the `-fPIC` arguments it both compiles correctly and functions identically to the PyPi package. Hope this fix gets pushed into the repo as then it'll add compatibility for Ubuntu Mate for other users.

Thanks again for getting this working!

Gadgetoid, 2015-09-24:
I've  now tested this on Raspbian and pushed the `-fPIC` fix into the Makefile: https://github.com/pimoroni/unicorn-hat/commit/52010b2eac166c78cd02906302bdc16a48e6403d

closed 2015-11-02 by Gadgetoid