pimoroni/unicorn-hat #35: rpi-ws281x broken on the latest ArchLinux

jibi, 2015-09-19:
With the latest ArchLinux release, running on an A+, unicornd is not working.
After digging a bit with gdb I found that the library stucks here

```c
    while (!(cm_pwm->ctl & CM_PWM_CTL_BUSY))
```

which is at `python/rpi-ws281x/lib/ws2811.c:261`.

bendo, 2015-12-02:
Last five commits should fix issue with ArchLinux and GCC >= 5.1.0
Sorry, that is not a one fix commit, but I made it in hurry direct on github.

mil, 2016-01-18:
Any word on this or the related PR @bendo? 
What needs to be done to run `unicornd` on Arch linux ARM? 

On your fork I can get `c/unicorn/unicorn.c` compiled and working but `unicornd` and the python test scripts don't work for me -- `unicornd` occasionally'll throw `DMA error` codes.

bendo, 2016-01-21:
@mil Back then, I made a quick fix. I can look at `unicornd` and python test scripts, but probably on weekend. No time these days.

mil, 2016-01-21:
@bendo can't wait -- most thanks for your help

Gadgetoid, 2016-07-12:
I'm labeling this as such so this issue is open for future Arch users, but not on my "radar" so to speak.

A PR which fixes Arch compatibility without breaking Raspbian, which is my primary concern, would be very welcome. I really don't have the time to work on it myself :(

Gadgetoid, 2016-10-11:
What's the current state of building on Arch? It might be worth submitting a patch upstream, since that's where I inherit the rpi_ws281x code from, and I plan to completely remove rpi_ws281x from this repository as soon as possible, to rely on the original version.

https://github.com/jgarff/rpi_ws281x

bendo, 2016-10-11:
@Gadgetoid I think my fork should work fine. To switch rpi_ws281x to original version is good idea. I wanted to do it as well in my fork, but there is no time... Two little kids. :-) You can do changes which you want, and when you will be ready I can try it on Arch. If there will be needed some correction, I will send you pull request.

matthewbullweb, 2016-11-06:
I can't seem to install this either. This is the last error message I get.

Gadgetoid, 2017-02-07:
As mentioned above, but with some progress now, I plan to switch over to https://github.com/pimoroni/rpi_ws281x-python soon, which includes the library directly from the upstream dev branch (with PCM and SPI support for driving LEDs) which can be found here: https://github.com/richardghirst/rpi_ws281x

closed 2017-02-07 by Gadgetoid