pimoroni/unicorn-hat #48: Simple example for Arduino

mralexgray, 2016-02-27:
I know it is possible... but I have been too dumb to figure out what is needed (libraries, etc) / how to connect the hat - to an Arduino - for simple usage.

Any example / help would be much appreciated!

Gadgetoid, 2016-03-21:
You can use Adafruit's NeoPixel library to drive UnicornHAT from an Arduino. Use the VCC, DIN and GND pins on the edge of UnicornHAT to connect to 5V, a data pin, and Ground respectively and you're good to go: https://learn.adafruit.com/adafruit-neopixel-uberguide/arduino-library

closed 2016-03-30 by Gadgetoid

xTauv, 2017-10-28:
Gadgetoid, I'm really new to this. So I want to attach the UnicornHAT to an Arduino Uno. Could you simplify the adafruit NeoPixel code for me? I'm very confused.
- Thanks!

xTauv, 2017-10-29:
[image]

Should I use the NeoMatrix library instead?

Gadgetoid, 2017-10-29:
Looks like you're wiring the right pins to the right connections on the HAT, but without a more robust connection you'll probably find it doesn't work. Since the ends of those wires are square, and the holes on the PCB are round, the contact surface is at best negligible.

You can either:

a. Solder the wires, or a female header to the Unicorn HAT to make a better connection
or
b. Plug the wires directly into the relevant connections on the 40-pin header. See: https://pinout.xyz/pinout/unicorn_hat which gives a top down view of which pins to use (looking from the top of the HAT)

Then to init the library on Arduino:

```arduino
#include <Adafruit_NeoPixel.h>
#define PIN 6
Adafruit_NeoPixel strip = Adafruit_NeoPixel(64, PIN, NEO_GRB + NEO_KHZ800);
void setup() {
  strip.begin();
  strip.show(); // Initialize all pixels to 'off'
  strip.setPixelColor(11, 255, 0, 255); 
  strip.show();
}
void loop() {
  // put your main code here, to run repeatedly:
}
```