pimoroni/unicorn-hat #55: ws2811_init failed with code -1

0x13io, 2016-05-13:
I have recently wiped my Raspi (model b 2) and I'm now unable to use my unicorn hat.

Using the matrix example running as sudo the following happens:

> pi@raspberrypi:~/Pimoroni/unicornhat $ sudo ./matrix.py 
Traceback (most recent call last):
  File "./matrix.py", line 3, in <module>
    import unicornhat as unicorn
  File "/usr/local/lib/python2.7/dist-packages/unicornhat.py", line 14, in <module>
    ws2812.begin()
  File "/usr/local/lib/python2.7/dist-packages/neopixel.py", line 114, in begin
    raise RuntimeError('ws2811_init failed with code {0}'.format(resp))
RuntimeError: ws2811_init failed with code -1
Exiting cleanly

Gadgetoid, 2016-05-17:
This is some weird problem with revision number. It should now be fixed, but let me know if otherwise.

To update to the latest version of the library, run:

```
sudo pip install -I rpi_ws281x
```

closed 2016-07-12 by Gadgetoid

strohrbaugh, 2017-05-09:
Fast forward a year, with the 3/20/2017 distribution of Jessie freshly installed, and I'm getting the exact same issue, same output, using the updated version of the library. Any idea where to get started?

djungowski, 2017-06-22:
I was having the exact same issue until I realized that I simply forgot to run my python scripts as root.