pimoroni/unicorn-hat #56: Garbled/Intermittent display on NOOBS 1.9.1 / Raspbian 2016-05-10 or newer

Gadgetoid, 2016-05-17:
On Raspbian 2016-05-10 / NOOBS 1.9.1 booting to X seems to do something with analog audio, where it would have otherwise been left alone and worked, mostly, by sheer luck. You need to *force* your Pi to HDMI audio, thus leaving the PWM hardware that analog audio normally uses to Unicorn HAT, using these lines in `/boot/config.txt`

```
hdmi_force_hotplug=1
hdmi_force_edid_audio=1
```

( Use `sudo nano /boot/config.txt` to edit the file )

This pretends an HDMI monitor is plugged in, and that it supports audio.

You can also start in CLI mode, which doesn't load whatever audio stuff is being used in X and has the same effect.

lurch, 2016-05-19:
Interesting... is it worth reporting this against https://github.com/RPi-Distro/repo/issues ?

RogueM, 2016-05-23:
I had a hell of a time reproducing until I tried Raspbian as opposed to NOOBS.
It seems to be because NOOBS 1.9.1 implements the following in /boot/config.txt

`hdmi_force_hotplug=1`

which seems to be enough to leave PWM working until you start using analog audio, for example a simple speaker-test is enough to mess things up until the next reboot.

... I guess my comment has limited value, but I thought it might at least narrow down the appropriate environment to produce the issue out of the box, consistently.

JezShed, 2016-07-04:
I have recently experienced this exact problem when updating & upgrading my Pi-Top. It runs a slightly customised Rasbian and has been working flawlessly with the Unicorn Hat until last week when I performed the update. The problem persisted even when audio in the GUI was set to HDMI.

Forcing the Raspi to use the HDMI audio in the config.txt file however did solve it:
```
hdmi_force_hotplug=1
hdmi_force_edid_audio=1
```
Thanks Gadgetoid for the solution.

Regards,
Jez

closed 2017-03-16 by RogueM