pimoroni/unicorn-hat #57: get / set pixels function

lurch, 2016-05-19:
I just had a look at https://github.com/pimoroni/unicorn-hat/blob/master/python/UnicornHat/unicornhat.py and I wonder if the `set_pixels` and `get_pixels` functions need to be updated to use the `get_shape` function?

Or is it expected that the PHAT will always simply ignore the "bottom half" of the 8*8 block?

At the very least, the docstring for `get_pixels` ought to be updated to say "8x8x3" ;-)

lurch, 2016-05-19:
(And I also guess some of the demos might need updating too?)

dglaude, 2016-05-26:
I have updated most of the example to be pHAT/HAT compatible using autodetect and get_shape.
(I have just made a pull request of my code, waiting for that you can look at [https://github.com/dglaude/unicorn-hat](url) )

Not sure about the change you want on set_pixels and get_pixels, I did not touch that part.

Gadgetoid, 2016-07-12:
The revised version of this PR has been merged now, and I can see where the change might come in handy. Both `get_pixels` and `set_pixels` have hard-coded values of 8 and 8 for width/height, which is incorrect for a pHAT:

```python
def get_pixel(x, y):
    """Get the RGB value of a single pixel"""
    index = get_index_from_xy(x, y)
    if index is not None:
        pixel = ws2812.getPixelColorRGB(index)
        return int(pixel.r), int(pixel.g), int(pixel.b)

def set_pixels(pixels):
    for x in range(8):
        for y in range(8):
            r, g, b = pixels[y][x]
            set_pixel(x, y, r, g, b)
```

Gadgetoid, 2016-10-11:
I think I've fixed this in: cf9ba6f8feaf25038895d625bd50b681c1b47c0f

Plus corrected the fact that `set_all` does *not* in fact work like a pixel shader, added `shade_all` which does, and made `set_all` just call `shade_all` with a lambda to pull out the right colours from the 2d array.

closed 2016-10-11 by Gadgetoid