pimoroni/unicorn-hat #67: Examples Not Installed

MisterEmm, 2016-07-05:
Hi I'm having problems installing this for my Unicorn pHat on a Pi 3, fresh install of Raspbian. The "Simple" installation goes OK until it gets to the examples section, then it throws the error as follows:

Successfully installed unicornhat rpi-ws281x
Cleaning up...
Done!

Downloading Unicorn HAT/pHAT examples...
Cloning into 'unicorn-hat'...
remote: Counting objects: 130, done.
remote: Compressing objects: 100% (115/115), done.
remote: Total 130 (delta 20), reused 61 (delta 10), pack-reused 0
Receiving objects: 100% (130/130), 327.46 KiB | 0 bytes/s, done.
Resolving deltas: 100% (20/20), done.
Checking connectivity... done.
**cp: cannot stat ‘/tmp/pimoroni.iGrsfE/unicorn-hat/python/examples//*’: No such file or directory**
Examples copied to /home/pi/Pimoroni/unicornhat/

-----------
The folders are there but the unicornhat subfolder is empty. I've tried it a couple of times with the same result, installed fine on a Pi 2 2 weeks ago.

Thanks!

MisterEmm, 2016-07-05:
I think it might be a difference between the setup script and the folder structure, the easy setup created a folder within Pimoroni called unicornhat whereas when I git cloned it just now it created one called unicorn-hat (i.e. with a dash in it).

Gadgetoid, 2016-07-05:
We're currently in the middle of restructuring the repositories to support deb-based distribution of our packages. Bear with us a moment :D

MisterEmm, 2016-07-05:
Thanks! I thought it'd be something like that, working fine otherwise, just spent 15 minutes playing with the paint thing, can't see properly now.

RogueM, 2016-07-05:
it should be fixed now!

MisterEmm, 2016-07-05:
Awesome thank you!

Martin

closed 2016-07-07 by Gadgetoid