pimoroni/unicorn-hat #70: What to do if you see: ws2811_init failed with code -1

Gadgetoid, 2016-07-12:
If you see the `ws2811_init failed with code -1` message, a number of things could be going wrong since the error -1 is returned for *all* exceptions when initializing the ws2811 library.

The most common cause thus far, however, is a new, or unknown revision of Raspberry Pi which is not logged in the `rpihw.c` file. To determine your revision number run:

```
curl /proc/cpuinfo | grep Revision
```

Compare this against the numbers in: https://github.com/pimoroni/unicorn-hat/blob/master/library/rpi-ws281x/lib/rpihw.c

If you don't see your revision number there, let us know. You should also submit an issue against https://github.com/jgarff/rpi_ws281x, with details of your Pi.

macom2000, 2016-12-21:
`cat /proc/cpuinfo | grep Revision`

gives 

`Revision : 0000`
What can I do?

tolgraven, 2017-01-27:
Zero v1.3
```
CPU revision	: 7
Revision	: 1900093
```
Building and running rpi_ws281x lib straight actually works, but unicorn fails with -1.

martinsmac, 2017-03-04:
I am having this problem on a Picture Zero W. 
9000c1
You have defined this board in the header.  Any ideas?

JustinGT, 2017-03-05:
Same issue here on Pi Zero W (with latest apt-get update etc).
Revision        : 9000c1
Which is defined in the header: https://github.com/pimoroni/unicorn-hat/blob/master/library/rpi-ws281x/lib/rpihw.c#L205
Has everything been recompiled or is this something else?

martinsmac, 2017-03-05:
I rebuilt the rpi_ws2811 library on my pi w and it works fine now.

RogueM, 2017-03-05:
interesting... recompiled from the repo you mean, right? hum, yes, we have been thinking it's the pypi distributed source that somehow is dodgy in some way, though at a quick glance it wasn't obvious. Certainly the source in this repo is what we tested against the zero W.

martinsmac, 2017-03-05:
I cloned this repo and recompiled the C library locally.  I then ran setup.py.  My intention was simply to add better error messages, rather than -1 for all cases.  Once recompiled it started working and I never saw the errors codes I had added.

djungowski, 2017-06-22:
I was having the exact same issue until I realized that I simply forgot to run my python scripts as root.

closed 2017-11-20 by Gadgetoid

juan-Chargevite, 2021-08-03:
cat /proc/cpuinfo | grep Revision
shows:
Revision        : 0000
My hardware is a NanoPi neo Core
Can you tell me how to solve it?

Gadgetoid, 2021-08-03:
The NanoPi neo Core is not a Raspberry Pi board and will not work with rpi_ws281x. You'd need to figure out how to exploit its hardware features to generate a WS281X-compatible signal, or get Googling and see if someone else has done so. While rpi_ws281x is extremely specific to the Pi hardware, the techniques it uses are likely portable by someone with the right documentation, expertise and inclination.

Looks like there's been interest in ws2812 on the NanoPi since at least 2016 with nobody rising to the challenge, so YMMV - http://www.friendlyarm.net/forum/topic/6839

The easy solution? Buy a Raspberry Pi.