pimoroni/unicorn-hat #79: Ambiguity with snow example

r41d, 2016-07-22:
In the snow example line 36 says
```python
row[randint(0, width - 1)] = 50 + randint(0, 155)
```
which yields a value between 50 and 205.

The comment in line 46 says
```python
# val is between 50 and 255
```
but the maximum value that could come from the above calculation is 205.

Suggestion:
- Either raise 155 to 205 (line 36)
- or lower 255 to 205 (line 46)

Gadgetoid, 2016-07-24:
Good spot, and a good case against comments like this which don't really add any value to what they're commenting. I'd go with option three; delete the comment. Or maybe swap it for something a bit more useful ;D

closed 2016-07-24 by Gadgetoid

r41d, 2016-07-24:
Well, now it says at the top
```py
MIN_BRIGHTNESS = 50
MAX_BRIGHTNESS = 155
```
but the maximal possible brightness value from
```py
MIN_BRIGHTNESS + randint(0, MAX_BRIGHTNESS)
```
is 205, but the reader of the code would think that it should be 155 (line 8).

Good job, you made it worse.

Gadgetoid, 2016-07-24:
Well that'll teach me to make drive-by changes over the weekend :D 

You're right, `MAX_BRIGHTNESS` should be 205 and the calculation should be:

```
MIN_BRIGHTNESS + randint(0, MAX_BRIGHTNESS - MIN_BRIGHTNESS)
```