pimoroni/unicorn-hat #84: unicornd broken?

SBejga, 2016-09-05:
Is somehow the unicornd broken? Can not get it working now.
I have installed the unicornd deamon and started it. But the test clients won't work. Will stay black.
If I run the normal python examples or the unicorn binary the leds will work. But only if I start the example twice. First run will also show nothing. The second run will show then the example (e.g. simple.py)

Any guess what could be the reason for unicornd stop working?

Gadgetoid, 2016-11-07:
I guess it's just fallen into disrepair from being unmaintained, and probably doesn't support either recent hardware revisions or kernel changes.

Might be worth summoning @jibi to see if he fancies fixing it ;D

Gadgetoid, 2017-09-14:
I've just updated the build paths and tested/run `unicornd`, it seems okay. It might have ended up linked against a hopelessly obsolete clone of `rpi_ws281x` in this repository that I have since removed.

It now pulls down from our up-to-date mirror/fork. See updated build instructions.

closed 2017-09-14 by Gadgetoid