pimoroni/unicorn-hat #85: It's normal

francoismartin, 2016-09-12:
Hi,

I bought unicorn phat and I installed in my raspi 2.

When I launch simple.py. I have already 3 leds lighted 

[image]

Normally the code show all led, no ?

And when I write :

```
unicorn.set_layout(unicorn.PHAT)
unicorn.brightness(0.8)
unicorn.rotation(0)

unicorn.set_pixel(0, 0, 255, 0, 0)

unicorn.show()
```

I have 

[image]

We are oki, it's not normal ?

Thanks

Pyroseza, 2016-09-12:
Do you similar behaviour when you run the examples?

Kind Regards,
Jarrod Price

--------

francoismartin, 2016-09-12:
When I runed `rainbow_blink` I had nothing

This video is when I runed `drop.py`

[image]

RogueM, 2016-09-12:
Have you run the one-line installe? This will ensure that the audio does not interfere with the PWM signal sent to GPIO18 to control the Unicorn.

Pyroseza, 2016-09-12:
Here is the one liner:

curl -sS get.pimoroni.com/unicornhat | bash

francoismartin, 2016-09-13:
@RogueM and @Pyroseza 

Thanks.
The first time I had installed with the one liner but It wasn't work but now. It work.

Thank you very much.

closed 2016-09-13 by francoismartin