pimoroni/unicorn-hat #86: How can I to keep the led lighted ?

francoismartin, 2016-09-13:
Hi,

I would like to turn on the led with command and turn off the led with another command

```
#!/usr/bin/env python

import argparse
import unicornhat as unicorn

unicorn.set_layout(unicorn.PHAT)
unicorn.brightness(0.2)

parser = argparse.ArgumentParser()

parser.add_argument("-sw", "--show", action="store_true", help="turn on the led")
parser.add_argument("-st", "--stop", action="store_true", help="stop the led")

args = parser.parse_args()

if args.show:
    for x in range(8):
        unicorn.set_pixel(x, 1, 255, 0, 255)
    unicorn.show()
    
elif args.stop:
    unicorn.off()
```

But when this script is executed (it's work) the leds turn off automaticly when the script python is finish `Exiting cleanly` 

How can I to keep the led lighted ?

Thanks

Pyroseza, 2016-09-13:
The best way I can think of is to write a script that does not exit until you tell it to. So you would have to write a script that loops awaiting input from the keyboard or whatever other interface you want to implement, then when you have a certain key pressed you can make it light up the LEDs and then while the LEDs are lit you can make it loop awaiting further input. So that when you then press another key it then either turns the LEDs off pending further input to turn it on again or it can terminate the script entirely.

I am planning on writing a script like this soon for my own purposes I can share with you when I have something working.

Kind Regards,
Jarrod Price

--------

francoismartin, 2016-09-13:
oki thanks @Pyroseza

closed 2016-09-13 by francoismartin

Pyroseza, 2016-09-13:
Here you go, try this.
You just need to paste it in a clean script and make sure that the format is correct. I'm using my phone so really sorry for not uploading an actual script file, it does work perfectly fine on the phat (4x8), let me know if there are any issues.
I'll see if I can fork the actual main branch and commit some of my samples, possibly for everyone to use? I dunno... depends on the interest I guess.

#!/usr/bin/env python

import unicornhat as unicorn
import time

unicorn.set_layout(unicorn.PHAT)
unicorn.brightness(0.5)

bToggle = True

print("Unicorn PHat Toggle LEDs with input!")
time.sleep(.5)
print("Press <enter> to toggle or <ctrl> + <c> to quit")
time.sleep(.5)

def toggle():
   global bToggle
    print("On!" if bToggle else "Off!")
    i = 255 if bToggle else 0
   bToggle = not bToggle
    for y in range(4):
        for x in range(8):
            unicorn.set_pixel(x,y,i,i,i)
   unicorn.show()

while True:
    toggle()
    raw_input("")

Kind Regards,
Jarrod Price

--------

Pyroseza, 2016-09-15:
Here you go, much better version of the code and it randomises the colours:

https://github.com/Pyroseza/unicorn-hat/blob/master/examples/phat/Toggle.py

RobinCutshaw, 2020-05-05:
I resolved it with this:

```
import atexit
import signal

def cleanup():
    print("cleanup!")
    unicorn.show()
    if signal_on_cleanup:
        os.kill(0, signal.SIGKILL)

signal_on_cleanup = False
try: atexit.unregister(unicorn._clean_shutdown)
except AttributeError: signal_on_cleanup = True
atexit.register(cleanup)
```

The unicornhat library registers _clean_shutdown via atexit which then calls off().  You have to deregister it (python3) or send a SIGKILL (python2 since there is no atexit.unregister).

Robin