pimoroni/unicorn-hat #94: overloaded methods for colour tuples

PJOttCanHome, 2017-01-21:
I wrote some code for the Sense Hat that would be nice to port over.  Could you create overloaded methods that allow for colour tuples like was done for the sense hat (and allow clear() to accept (r,g,b) even though set_all() does.  Using a colour tuple is often preferable.
Thanks

Gadgetoid, 2017-02-08:
I've added support for this into the dev branch which I'm currently working on, if you could install it manually and give it a go I'd appreciate feedback!

https://github.com/pimoroni/unicorn-hat/tree/dev

PJOttCanHome, 2017-02-08:
Thanks, but it’ll take me a while to test it.  Somehow my Unicornhat just got hosed 2 days ago and so I ordered a new one yesterday.

 

https://github.com/pimoroni/unicorn-hat/tree/dev

  _____  

No virus found in this message.
Checked by AVG - www.avg.com <http://www.avg.com/email-signature> 
Version: 2016.0.7998 / Virus Database: 4756/13870 - Release Date: 01/31/17
Internal Virus Database is out of date.

closed 2017-11-20 by Gadgetoid