pimoroni/unicorn-hat #96: Note about disabling the sound card

pambrose, 2017-03-03:
The README.md needs to highlight the fact that the Unicorn Hat requires the Raspi sound card to be disabled.  You can find numerous people on the Pimoroni msg board that learned this lesson the hard way.

RogueM, 2017-03-05:
there is ni reason to disable on-board audio. What is required is to forcefully route audio to hdmi, by enabling `hdmi_force_hotplug=1`.. well you CAN disable audio altogether, but it is like using a sledgehammer to put a picture frame up.

pambrose, 2017-03-05:
There are a bunch of people on your msg board that would argue otherwise. Regardless, the README.md should say more about the audio PWM issue. Simply running pip during installation is not sufficient and the README.md should say so.

RogueM, 2017-03-05:
I don't disagree with you that the ReadMe can be improved, I am just trying to put the myth that the audio card needs disabling to rest, it isn't correct and I keep seeing this popping up as a fact.

... I'll prod @Gadgetoid to add an appropriate notice as and when he gets a moment.

closed 2017-03-16 by RogueM