pimoroni/unicorn-hat #97: ws2811_init failed with code -1 (but works after a restart!)

mikebaldry, 2017-04-11:
We are seeing this error after a few days running - Everything works fine, and then it starts getting this error and the only way to get it working again is to restart the Pi.

`Revision	: a21041`

What could be happening?

mikebaldry, 2017-04-11:
I feel like it could be to do with us killing the process periodically and restarting it - is there some way we have to release ws2811 that we aren't doing?

Gadgetoid, 2017-04-11:
It's possible but with the current version of the library, it's virtually impossible to know what's actually going on since *everything* fails with code `-1`.

The soon-to-release version will produce specific error codes highlighting exactly what's at fault, which should let us get to the bottom of this quickly.

mikebaldry, 2017-04-11:
Ok, is there any way I can test the unreleased version now?

Gadgetoid, 2017-06-28:
Slightly late post, but absolutely. You can grab the dev version of the Unicorn HAT library from: https://github.com/pimoroni/unicorn-hat/tree/dev/library/UnicornHat

And pair it with the new version of rpi-ws281x-python: https://github.com/pimoroni/rpi_ws281x-python

```
git clone https://github.com/pimoroni/rpi_ws281x-python
cd rpi_ws281x-python
git submodule update --init
cd library
sudo python setup.py install
```

```
git clone https://github.com/pimoroni/unicorn-hat -b dev
cd unicorn-hat/library/UnicornHat
sudo python setup.py install
```

closed 2017-06-28 by Gadgetoid