pimoroni/unicorn-hat #98: Script stops after a few seconds when started on boot

BoldBigflank, 2017-06-11:
I made a little script that shows a different color based on the hour of the day. When I run it by hand it runs properly. When I hook it up to a crontab or put it in rc.local, it goes for a couple seconds, then stops. I'm not really sure what causes it, but I do know that both crontab and rc.local have the same effect.

```
#!/usr/bin/env python
# @reboot /home/pi/clock.py >> /home/pi/clock.log 2>&1

import math
import time
import datetime
# datetime.datetime.now()

import unicornhat as unicorn

unicorn.set_layout(unicorn.AUTO)
unicorn.rotation(0)
unicorn.brightness(0.5)
width,height=unicorn.get_shape()

def mainLoop():
    i = 0.0;
    while True:
        current_time = datetime.datetime.now()
        current_hour = current_time.hour
        # orange, red for nighttime
        # blue, green for awake time
        if(current_hour <= 6):
            setLight(255, 0, 0, 0.2) # Low red
        elif(current_hour == 7):
            setLight(255, 128, 0, 0.2) # orange
        elif(current_hour == 8):
            setLight(65, 105, 225, (current_time.minute / 180.0)+0.2) # royal blue
        elif(current_hour > 20):
            setLight(255, 0, 0, 0.2) # Low red
        else:
            setLight(138, 43, 226, 0.5)
        i = i + 0.3
        unicorn.set_pixel(0,0,int(math.cos(i) * 64.0 + 128.0), 0, 0)
        time.sleep(0.1)

def setLight(r,g,b,intensity):
    unicorn.brightness(intensity)
    for y in range(height):
        for x in range(width):
            if(x == 0 and y == 0): # control pixel
                continue
            unicorn.set_pixel(x,y,int(r),int(g),int(b))
    unicorn.show()

mainLoop()
```

Pyroseza, 2017-06-12:
Any errors in the log or is it just abruptly ending?

You should also be able to look in the system messages.

BoldBigflank, 2017-06-12:
I get no errors when I set up a log file on the cron task, it just stops. You can see I set the pixel at 0,0 to do a "breathing" red, so I can tell it breathes for 2-3 seconds and then freezes on one brightness.

Pyroseza, 2017-06-13:
Odd I can have a look when I get a chance, not sure if the guys from Pimoroni have seen this yet.

I don't see you setting pixel 0,0, from the looks of it you skip 0,0 entirely.

Also just out another question related to the script in general.

It loops every 0.1s but only changes colours on different hours. That's 36000 iterations an hour with nothing else in between except setting one colour over and over, and the chance of it performing another action is once an hour.

Why not make it a bit more efficent and sleep for 60s or more or calculate how long to sleep for until the next hour change?

Just looking at it from a sys admin POV.

Pyroseza, 2017-06-13:
One more thing, your low red is the brightest red. Maybe you want something more like 128,0,0?
I have a breathing script sitting somewhere I'll throw it over to you when I get to my pi

BoldBigflank, 2017-06-13:
I set 0,0 in the while loop just before sleep. This does need to be updated often to get the "breathing" right. Besides, this is all my zero is doing, so 10 times a second is not stressing the device at all. I can optimize after it is working.

Ah, the low red is fully red, but has a low intensity. I could cut that down to make it dimmer, for sure.

Pyroseza, 2017-06-13:
Ah sorry there it is! I did not see it :)

Pyroseza, 2017-06-13:
@BoldBigflank right so I fiddled around a for a few minutes now and it's quite simple to get it working without crashing, I found it crashing on trying to access /dev/mem so you need to change your crontab to run it with "sudo python" this way it works fine for me and does not terminate at all.

Here I've saved it as timecolour.py:

    #make it run at a specific time every day -> 9:15pm
    15 21 * * * sudo python ~/scripts/timecolour.py 1>~/scripts/timecolour.log 2>&1 &

    #make it run on restart
    @restart sudo python ~/scripts/timecolour.py 1>~/scripts/timecolour.log 2>&1 &

Pyroseza, 2017-06-13:
@BoldBigflank another thing to take note of is that you cannot terminate it cleanly, so if you do just "sudo kill <pid>" it then it will leave the LEDs glowing so maybe make it look for a token and then it can exit gracefully.

BoldBigflank, 2017-06-14:
I had been using root's crontab, so this permissions thing doesn't seem to be the same issue I am seeing.

Pyroseza, 2017-06-14:
Try adding the extra ampersand right at the end.

I'm not with my pi again, did not think to try it with root. What if you do it as a normal user with sudo?

BoldBigflank, 2017-06-14:
I tried setting up the crontab like you suggested but with the example rainbow.py with the same result, 2-5 seconds of animation then frozen.

Pyroseza, 2017-06-14:
And what about if you run it as a normal user with sudo, does that do the
same or does it work?

BoldBigflank, 2017-06-14:
The behavior is the same with sudo.

--Alex Swan
www.bold-it.com

Pyroseza, 2017-06-14:
Okay I've tried all sorts of ways to break this but it works every time for me.
You sure that there are no messages in the system log (/var/log/messages) or in your mail (/var/mail)?
What python do you have installed?
What shell are you using?

Gadgetoid, 2017-09-14:
Did you folks ever get to the bottom of this? Sorry for not chiming in, but I had no more insight to offer!

Pyroseza, 2017-09-14:
Nope I can't replicate this, never got more info either

closed 2017-11-20 by Gadgetoid