pimoroni/unicorn-hat #99: Building rpi-ws281x fails on arch linux

4goettma, 2017-06-27:
I'm using Arch Linux ARM on my Raspberry Pi 3, about a year ago I used bendo's fork with the instructions from [0]. Everything worked. Now I didn't just his fork [1] but also [2] and "pip install rpi_ws281x", but all of them fail while compiling with the error message "/usr/bin/ld: cannot find -lws2811". I already tried to adjust some of the include paths but wasn't able to resolve it. The default python version ("python") in Arch Linux is Python 3.

[0]: http://forums.pimoroni.com/t/unicorn-hat-on-arch-linux-yet-another-thread/2090/9
[1]: https://github.com/bendo/unicorn-hat
[2]: https://github.com/pimoroni/unicorn-hat
```
Linux alarmpi 4.9.34-1-ARCH #1 SMP Mon Jun 26 01:32:24 UTC 2017 armv7l GNU/Linux
```
```
[alarm@alarmpi rpi-ws281x]$ gcc --version
gcc (GCC) 7.1.1 20170516
```
Using the official repo:
```
[alarm@alarmpi rpi-ws281x]$ sudo make
sudo rm -rf build
sudo ./setup.py build
running build
running build_py
Compiling ws281x library...
creating build
creating build/lib.linux-armv7l-3.6
copying neopixel.py -> build/lib.linux-armv7l-3.6
running build_ext
building '_rpi_ws281x' extension
creating build/temp.linux-armv7l-3.6
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -fPIC -Ilib/ -I/usr/include/python3.6m -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o
gcc -pthread -shared -Wl,-O1,--sort-common,--as-needed,-z,relro -Wl,-O1,--sort-common,--as-needed,-z,relro build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o -Llib/ -L/usr/lib -lws2811 -lpython3.6m -o build/lib.linux-armv7l-3.6/_rpi_ws281x.cpython-36m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lws2811
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1
make: *** [Makefile:9: build] Error 1
```
Using the pip:
```
[alarm@alarmpi rpi-ws281x]$ sudo pip install rpi_ws281x
Collecting rpi_ws281x
  Using cached rpi_ws281x-2.0.5.tar.gz
Installing collected packages: rpi-ws281x
  Running setup.py install for rpi-ws281x ... error
    Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ur01deto/rpi-ws281x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-f2_rqb8k-record/install-record.txt --single-version-externally-managed --compile:
    running install
    running build
    running build_py
    Compiling ws281x library...
    creating build
    creating build/lib.linux-armv7l-3.6
    copying neopixel.py -> build/lib.linux-armv7l-3.6
    running build_ext
    building '_rpi_ws281x' extension
    creating build/temp.linux-armv7l-3.6
    gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -fPIC -Ilib/ -I/usr/include/python3.6m -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o
    gcc -pthread -shared -Wl,-O1,--sort-common,--as-needed,-z,relro -Wl,-O1,--sort-common,--as-needed,-z,relro build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o -Llib/ -L/usr/lib -lws2811 -lpython3.6m -o build/lib.linux-armv7l-3.6/_rpi_ws281x.cpython-36m-arm-linux-gnueabihf.so
    /usr/bin/ld: cannot find -lws2811
    collect2: error: ld returned 1 exit status
    error: command 'gcc' failed with exit status 1
    
    ----------------------------------------
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-ur01deto/rpi-ws281x/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-f2_rqb8k-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-ur01deto/rpi-ws281x/
```

Gadgetoid, 2017-06-28:
I'm in the process of trying to replace the mess that exists in this Unicorn HAT repository with a properly separated out version of rpi_ws281x and, while we don't officially support Arch Linux, it probably couldn't hurt for me to look into this. Could you see if this problem happens with the new version?

Try grabbing: https://github.com/pimoroni/rpi_ws281x-python

```
git clone https://github.com/pimoroni/rpi_ws281x-python
cd rpi_ws281x-python
git submodule update --init
cd library
python setup.py build
```

Gadgetoid, 2017-06-28:
Okay, as of https://github.com/pimoroni/rpi_ws281x/pull/4 the linked repository *should* build on Arch Linux. That is to say; I've tested it, and it builds for me. It looks like the library compile was hard failing by treating a deprecation warning as an error. It had already been fixed upstream, so I merged it down.

Now need to test and make sure I haven't broken anything!

4goettma, 2017-06-28:
Doesn't change anything, because I'm fetching from the wrong source.
```warning: redirecting to https://github.com/pimoroni/rpi_ws281x/```

Full listing:
```
[alarm@alarmpi ~]$ git clone https://github.com/pimoroni/rpi_ws281x-python
Cloning into 'rpi_ws281x-python'...
remote: Counting objects: 113, done.
remote: Total 113 (delta 0), reused 0 (delta 0), pack-reused 113
Receiving objects: 100% (113/113), 64.86 KiB | 61.00 KiB/s, done.
Resolving deltas: 100% (61/61), done.
[alarm@alarmpi ~]$ cd rpi_ws281x-python
[alarm@alarmpi rpi_ws281x-python]$ git submodule update --init
Submodule 'library/lib' (http://github.com/pimoroni/rpi_ws281x) registered for path 'library/lib'
Cloning into '/home/alarm/rpi_ws281x-python/library/lib'...
warning: redirecting to https://github.com/pimoroni/rpi_ws281x/
Submodule path 'library/lib': checked out 'bb8a603a08b243b219b5decc4577f265a04b7735'
[alarm@alarmpi rpi_ws281x-python]$ cd library
[alarm@alarmpi library]$ python setup.py build
running build
running build_py
Compiling ws281x library...
b'lib/mailbox.c: In function \'mbox_open\':\nlib/mailbox.c:275:13: error: In the GNU C Library, "makedev" is defined\n by <sys/sysmacros.h>. For historical compatibility, it is\n currently defined by <sys/types.h> as well, but we plan to\n remove this soon. To use "makedev", include <sys/sysmacros.h>\n directly. If you did not intend to use a system-defined macro\n "makedev", you should undefine it after including <sys/types.h>. [-Werror]\n     if (mknod(filename, S_IFCHR|0600, makedev(100, 0)) < 0) {\n             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                                                                                                                                                                                             \ncc1: all warnings being treated as errors\nmake: *** [Makefile:17: lib-built/mailbox.o] Error 1\n'
creating build
creating build/lib.linux-armv7l-3.6
creating build/lib.linux-armv7l-3.6/neopixel
copying neopixel/__init__.py -> build/lib.linux-armv7l-3.6/neopixel
creating build/lib.linux-armv7l-3.6/rpi_ws281x
copying rpi_ws281x/__init__.py -> build/lib.linux-armv7l-3.6/rpi_ws281x
copying rpi_ws281x/rpi_ws281x.py -> build/lib.linux-armv7l-3.6/rpi_ws281x
running build_ext
building '_rpi_ws281x' extension
creating build/temp.linux-armv7l-3.6
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -march=armv7-a -mfloat-abi=hard -mfpu=vfpv3-d16 -O2 -pipe -fstack-protector-strong --param=ssp-buffer-size=4 -fPIC -Ilib/ -I/usr/include/python3.6m -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o
gcc -pthread -shared -Wl,-O1,--sort-common,--as-needed,-z,relro -Wl,-O1,--sort-common,--as-needed,-z,relro build/temp.linux-armv7l-3.6/rpi_ws281x_wrap.o -Llib-built/ -L/usr/lib -lws2811 -lpython3.6m -o build/lib.linux-armv7l-3.6/_rpi_ws281x.cpython-36m-arm-linux-gnueabihf.so
/usr/bin/ld: cannot find -lws2811
collect2: error: ld returned 1 exit status
error: command 'gcc' failed with exit status 1
```

For the moment I'm using an very old backup [0] I found on my found harddisk (already compiled), so I just have to use "sudo ./setup.py install". Working, but not a real solution. I think I cloned it back in the days from bendo's repository (not 100% sure).
Quite interesting is that compiling this **old backup works fine (compiles!)** on the Raspberry Pi 3 but fails to build (the usual error) on Arch Linux x86-64. I suppose it could fail due to gcc package updates etc. Ask if you need exact version numbers of involved packages.

[0]: https://share.riseup.net/#q5h32Io76U8ihwhODzrkbw

Gadgetoid, 2017-06-28:
Trash your local copy of the repo and run the steps above again. It's *definitely* working for me on a brand new install of Arch which I've just imaged and tested against.

I haven't installed anything additional but: python, python-pip, git, make, gcc

4goettma, 2017-06-28:
Excuse me, but  I'm a little bit confused.
- Which one did you clone? ```git clone --branch dev ...```? "steps above" refers to which post?
- ```[alarm@alarmpi library]$ python setup.py build``` Where are you? I just see a ```unicorn-hat/library``` directory, but there's no ```setup.py``` I could run.

Gadgetoid, 2017-06-30:
Sorry the setup.py is in library/UnicornHat. See: https://github.com/pimoroni/unicorn-hat/tree/dev/library/UnicornHat

emoryy, 2017-08-05:
Hi! What is the state of this? Is the current `dev` branch stiill the one that should be good on Arch?

I cloned this repo, changed to the `dev` branch, `cd` into `library/UnicornHat`, `sudo ./setup.py install`, and I get the same `/usr/bin/ld: cannot find -lws2811` error.

I also tried the steps from [this comment of yours](https://github.com/pimoroni/unicorn-hat/issues/99#issuecomment-311606113)
`/usr/bin/ld: cannot find -lws2811` again.

emoryy, 2017-08-08:
Ok, I was able to figure it out, so to summarize, here is what I did:
1.
```
git clone https://github.com/pimoroni/rpi_ws281x-python
cd rpi_ws281x-python/
git submodule update --init
```
now modify at line 37 in `lib/mailbox.c` change
`#include <sys/types.h>`
to
`#include <sys/sysmacros.h>`
then:

```
python setup.py build
sudo python setup.py install
```

2.
(cd out from the previous git repo)
```
git clone https://github.com/pimoroni/unicorn-hat.git
cd unicorn-hat/
git checkout dev
cd library/UnicornHat/
python setup.py build
sudo python setup.py install
```

After this I was able to run the Pimoroni example scripts (but only with sudo)

Gadgetoid, 2017-09-14:
I believe this fix is now upstream and in our fork of rpi_ws281x, since Raspbian users started running into it.

See: https://github.com/pimoroni/rpi_ws281x/blob/master/mailbox.c#L40

closed 2017-09-14 by Gadgetoid