pimoroni/unicorn-hat #101: corrupt disk when running unicorn-hat examples

hagna, 2017-07-19:
When I run this my disk times out (error -110) and the filesystem is mounted read only.

`#!/usr/bin/env python

import math, sys
import time

import unicornhat as unicorn

print("""Rainbow

Displays a beautiful rainbow across your HAT/pHAT :D

If you're using a Unicorn HAT and only half the screen lights up, 
edit this example and  change 'unicorn.AUTO' to 'unicorn.HAT' below.
""")

unicorn.set_layout(unicorn.AUTO)
unicorn.rotation(0)
unicorn.brightness(0.5)
width,height=unicorn.get_shape()

print("Reticulating splines")
time.sleep(.5)
print("Enabled unicorn poop module!")
time.sleep(.5)
print("Pooping rainbows...")

elapsed = 0
if len(sys.argv) >= 2:
	elapsed = int(float(sys.argv[1]))
i = 0.0
offset = 30
start = time.time()
while True:
        i = i + 0.3
        for y in range(height):
                for x in range(width):
                        r = 0
                        g = 0
                        r = (math.cos((x+i)/2.0) + math.cos((y+i)/2.0)) * 64.0 + 128.0
                        g = (math.sin((x+i)/1.5) + math.sin((y+i)/2.0)) * 64.0 + 128.0
                        b = (math.sin((x+i)/2.0) + math.cos((y+i)/1.5)) * 64.0 + 128.0
                        r = max(0, min(255, r + offset))
                        g = max(0, min(255, g + offset))
                        b = max(0, min(255, b + offset))
                        unicorn.set_pixel(x,y,int(r),int(g),int(b))
        unicorn.show()
	if (time.time() - start) + elapsed >= 30*1:
		print "timeout"
		break
        time.sleep(0.01)`

Dmesg says this:
```[  131.687494] mmc0: sbc op 23 arg 0xa0 flags 0x15 - resp 00000900 00000000 00000000 00000000, err 0
[  131.687504] mmc0: cmd op 18 arg 0x541268 flags 0xb5 - resp 00000900 00000000 00000000 00000000, err 0
[  131.687511] mmc0: data blocks a0 blksz 200 - err 0
[  131.687521] mmc0: stop op 12 arg 0x0 flags 0x95 - resp 00000000 00000000 00000000 00000000, err 0
[  131.687525] mmc0: =========== REGISTER DUMP ===========
[  131.687530] mmc0: SDCMD  0x00000052
[  131.687535] mmc0: SDARG  0x00541268
[  131.687540] mmc0: SDTOUT 0x017d7840
[  131.687545] mmc0: SDCDIV 0x00000003
[  131.687549] mmc0: SDRSP0 0x00000900
[  131.687554] mmc0: SDRSP1 0x00001269
[  131.687558] mmc0: SDRSP2 0x7fffffff
[  131.687564] mmc0: SDRSP3 0x00024007
[  131.687568] mmc0: SDHSTS 0x00000001
[  131.687573] mmc0: SDVDD  0x00000001
[  131.687577] mmc0: SDEDM  0x00010902
[  131.687582] mmc0: SDHCFG 0x0000040e
[  131.687587] mmc0: SDHBCT 0x000001c0
[  131.687591] mmc0: SDHBLC 0x000000a0
[  131.687595] mmc0: ===========================================
[  131.687847] mmcblk0: error -110 transferring data, sector 5509736, nr 160, cmd response 0x900, card status 0xb00```

Gadgetoid, 2017-07-20:
What power supply are you using with your Pi?

hagna, 2017-07-20:
5v 2500mA power supply connected to a powered usb hub.

Pyroseza, 2017-07-20:
Can you run a strace from a second terminal and output to a file? 

strace -o <output_filename> -p <pid>

hagna, 2017-07-20:
Ok I ran this command:
```
# strace -o foo ./begin 
```

https://pastebin.com/raw/ujbeAsc5

Pyroseza, 2017-07-21:
It's not finding any of your python libs, is "begin" the exact script from the repo?

Can you do yhe following?
$which python

Gadgetoid, 2017-07-21:
What do you mean by:

> 5v 2500mA power supply connected to a powered usb hub.

Is the Pi powering from the USB hub which is powered by another supply? Or is the USB hub powered by a 5v 2500mA supply?

Either way, I'd highly recommend directly powering the Pi with an official adaptor, since USB is ostensibly limited to 500mA which is not sufficient. It's quite likely that your hub is not designed to supply much more than that without severe voltage drop.

It may also be prudent to verify that outputting just black pixels to Unicorn HAT - IE driving the display with a signal, but not actually lighting any of the LEDs - does or does not cause the same effect.

I would be surprised if this were a failure in the Unicorn HAT itself... but stranger things have happened so I wont discount it entirely!

Pyroseza, 2017-07-25:
I don't think it is a power issue, I'm not disputing it. I just think the strace is revealing that the unicornhat libs are not installed.

Extract:
stat64("/home/pi/unicornhat", 0x7eaec200) = -1 ENOENT (No such file or directory)
open("/home/pi/unicornhat.arm-linux-gnueabihf.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
open("/home/pi/unicornhat.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
open("/home/pi/unicornhatmodule.so", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)
open("/home/pi/unicornhat.py", O_RDONLY|O_LARGEFILE) = -1 ENOENT (No such file or directory)

Make sure it is installed correctly and you have the correct permissions on them.

Gadgetoid, 2017-11-20:
This was, I believe, due to a DMA channel clash between the RPi_ws281x library and the SD card in newer versions of Raspbian. See: https://github.com/jgarff/rpi_ws281x/issues/208

It should be fixed now.

closed 2017-11-20 by Gadgetoid