pimoroni/unicorn-hat #110: Python 3 doesn't work.

bfayers, 2018-02-24:
Using the curl setup in the README doesn't work on python3 and produced:
`Traceback (most recent call last):
  File "rainbow.py", line 6, in <module>
    import unicornhat as unicorn
  File "/usr/local/lib/python3.5/dist-packages/unicornhat.py", line 3, in <module>
    from rpi_ws281x import __version__ as __rpi_ws281x__, PixelStrip, Color
ImportError: cannot import name '__version__'
` when using `import unicornhat`

even a `sudo pip3 install unicornhat` doesn't work.

Gadgetoid, 2018-02-24:
This looks like the rpi_ws281x library is out of date and not being updated, what does: `pip3 freeze | grep rpi_ws281x` show?

bfayers, 2018-02-24:
@Gadgetoid Thanks for the suggestion, didn't see this until just now 

I did a `sudo pip3 uninstall unicornhat` and a `sudo pip3 uninstall rpi_ws281x` and then did the `\curl -sS https://get.pimoroni.com/unicornhat | bash` and now it works. Weird one, guess the rpi_ws281x just wasn't being updated.

closed 2018-02-24 by bfayers