pimoroni/unicorn-hat #129: Weather icons, show_png examples fail

ghalfacree, 2019-01-29:
Fully up-to-date Raspbian running on a Raspberry Pi Zero with original low-resolution Unicorn HAT. Examples installed via curl'd script with all "yes please" optional stuff selected. Many examples - like matrix.py - work fine, but a few don't.

Steps to Recreate
==============
Install and update Raspbian. Install and update Pimoroni Unicorn HAT software.

```bash
$ cd Pimoroni/unicornhat/examples/hat
$ sudo ./show_png.py weather-icons/icons/rain.png
Traceback (most recent call last):
  File "./show_png.py", line 26, in <module>
    unicorn.set_layout(unicorn.HAT)
AttributeError: 'module' object has no attribute 'set_layout'
$ cd weather-icons
$ sudo ./weather-icons.py loop

    *************************************************************
    animated weather icons for the unicorn-hat by LovebootCaptain
    *************************************************************
    
Traceback (most recent call last):
  File "./weather-icons.py", line 69, in <module>
    width, height = unicorn.get_shape()
AttributeError: 'module' object has no attribute 'get_shape'
```

Reinstalling the examples changes nothing; neither does installing them via git instead.

Any ideas?

ghalfacree, 2019-01-29:
Actually, to correct myself, ```Pimoroni/unicornhat/matrix.py``` works fine, but ```Pimoroni/unicornhat/examples/hat/matrix.py``` does not:

```bash
$ sudo examples/hat/matrix.py
Matrix

Follow the white rabbit...

Traceback (most recent call last):
  File "matrix.py", line 14, in <module>
    unicorn.set_layout(unicorn.HAT)
AttributeError: 'module' object has no attribute 'set_layout'
```

Diff:
```diff
diff matrix.py examples/hat/matrix.py
3,4d2
< import unicornhat as unicorn
< from random import randint
5a4,12
> from random import randint
>
> import unicornhat as unicorn
>
>
> print("""Matrix
>
> Follow the white rabbit...
> """)
7c14
< unicorn.brightness(0.20)
---
> unicorn.set_layout(unicorn.HAT)
8a16
> unicorn.brightness(0.5)
```

Gadgetoid, 2019-01-30:
Try reinstalling the library- this looks suspiciously like file corruption:

```
sudo pip install -I unicornhat
```

If memory serves, that will reinstall `unicornhat`, ignoring any existing installs.

ghalfacree, 2019-01-30:
@Gadgetoid You're not wrong: that's fixed it. Ta!

closed 2019-01-30 by ghalfacree

Gadgetoid, 2019-01-30:
Excellent! Happy hacking.