pimoroni/unicorn-hat #132: 8.8.8.8 is blocked by my ISP, www.google.com works fine

nigels-com, 2019-04-28:
The installer is pinging 8.8.8.8 to check internet connectivity.
However, my ISP appears to be blocking that address for DNS purposes.

```
$ ping -q -w 10 -c 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.

--- 8.8.8.8 ping statistics ---
3 packets transmitted, 0 received, +3 errors, 100% packet loss, time 2074ms
pipe 3
```

Changing the script to www.google.com as a workaround works:

```
$ ping -q -w 10 -c 1 www.google.com
PING www.google.com (172.217.167.68) 56(84) bytes of data.

--- www.google.com ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 16.286/16.286/16.286/0.000 ms
```

Gadgetoid, 2019-04-29:
Yikes. This is the first I've heard of an *isp* blocking 8.8.8.8. Usually it's a user choice to rid themselves of Google's watchful eye.

Thinking out loud, since I think there's a more significant change that needs to happen here -

I honestly wonder if we need this check at all, since it's very possible for this particular test to pass while fetching package lists or packages from apt will fail due to firewall constraints.

It might be worth instead running an `apt update` and checking the status, since we almost always do that later in the script without error trapping anyway.

nigels-com, 2019-04-29:
My guess is that [Exetel](https://www.exetel.com.au/) blocked to this due to potential performance concerns.  (Australia to North America or Europe is laggy) . I don't agree with with it, but I'm entirely happy with their service aside from this particular quirk. 

If I recall correctly I ran into the same issue installing the Blinkt! Python stuff as well.

Pyroseza, 2019-05-14:
just my 2 pence 
[image]

Pyroseza, 2019-05-14:
what about pinging 1.1.1.1?
That's a much better one to ping is it not?

Gadgetoid, 2020-07-31:
@Pyroseza will give that a try and see how many complaints I get, thanks 😆

closed 2020-07-31 by Gadgetoid

Pyroseza, 2020-07-31:
Lol! Nice one Phil, can't believe it's been open this long! Should not have any issues with 1.1.1.1, after all it's CloudFlare 🤣

Gadgetoid, 2020-08-01:
Well- yeah- when CloudFlare is down the internet is technically down too 🙄

henrym9, 2020-08-02:
Heya folks,  I've got to complain about the opposite problem.  My ISP in Chile uses 1.1.1.1 for some of their internal stuff despite being notified that that's not an acceptable practice, but the end result is that it is blocked.  I've had to give up on using Cloudflare's DNS service (their CDN works fine), but 8.8.8.8 and 8.8.4.4 are our only options.

Gadgetoid, 2020-08-03:
Apparently this is a difficult problem!

In the long term, better exception handling might be the sensible alternative to trying to check connectivity beforehand- since even a *working* connection might not indicate that apt, or any other commands will work.

In the short term, perhaps a test that fails over from 1.1.1.1 to 8.8.8.8 and fails outright if neither respond would satisfy most cases.

Gadgetoid, 2020-08-03:
Something like the following should- in theory- satisfy both cases:

```bash
check_network() {
        sudo ping -q -w 10 -c 1 1.1.1.1 | grep "received, 0" &> /dev/null
        if [ $? -eq 0 ]; then
                return 0
        else
                sudo ping -q -w 10 -c 1 8.8.8.8 | grep "received, 0" &> /dev/null
                if [ $? -eq 0 ]; then
                        return 0
                fi
        fi
        return 1
}
```

henrym9, 2020-08-03:
Maybe pinging github.com?  If that doesn't work, it indicates either no DNS, or no connectivity to the site you'll be pulling from; either way that would be a fatal error, right?