pimoroni/unicorn-hat #133: half screen

puba, 2019-07-02:
When running examples on UnicornHat, only one half of hat is active, the other one is unactive.
Tried running detect.py and its solutions, still only half of hat visible.

Gadgetoid, 2019-07-03:
This is usually because your Unicorn HAT is being recognised as a pHAT or, rather, not recognised as a HAT. Due to the way the ID EEPROM works you need to plug the HAT on while your Pi is powered-off, or reboot.

You can also edit the examples to force them to treat your board as a HAT.

Find the line:

```
unicorn.set_layout(unicorn.AUTO)
```

And change it to:

```
unicorn.set_layout(unicorn.HAT)
```

(Or omit it altogether, since the default layout is HAT)

puba, 2019-07-03:
First, thanks for your answer. Tried all your suggestions, but still only half of hat active. Seems like it is recognised as pHAT, no matter what I do. Tried a fresh install of Raspbian also ->same result. Super annoying, no clue why.

Gadgetoid, 2019-07-03:
What happens if you run something like https://github.com/pimoroni/unicorn-hat/blob/master/examples/hat/matrix.py?

Which half is working? Could you post a photo?

puba, 2019-07-03:
Only first four rows from top to botttom are active - > running matrix

[image]

Gadgetoid, 2019-07-03:
Well this is truly odd. I think your HAT has a dry joint or damaged pixel that just so happens to correlate with the top/bottom half of the display. I can't see anything in the picture to indicate what's up, but these problems are often hard to spot visually.

The order of pixels on the Unicorn HAT and Unicorn pHAT is different, so the straight lines shown in your picture show that the code is outputting the right data, but the HAT just isn't displaying it. I think it's faulty, and the fact it's exactly half of the screen is a coincidence.

It might be worth pressing on the top left or top right most LED in the dead area while an example is running, jut to see if it springs into life. In the mean time could you drop an email to support so we can get you a replacement?

puba, 2019-07-03:
OK, will do, thank you.

Gadgetoid, 2019-11-01:
Did you get this sorted? I'll close this issue for now, but feel free to re-open if it's still applicable.

closed 2019-11-01 by Gadgetoid