pimoroni/unicorn-hat #136: What is the licence of library_c/unicorn and library_c/unicornd?

jacobbramley, 2020-02-01:
I ask because I'd like to submit a pull request affecting both tools.

There are both MIT and GPLv2 licence files at the repository top level, but it's not clear how these apply. I noticed that library_c/unicornd/unicornd.c has a GPLv2 banner at the top, but this is the only example of this under library_c.

Gadgetoid, 2020-02-25:
`unicornd` is user-contributed code which I should probably have requested a re-license to MIT on to match with our house licensing.

`unicorn.c`- is licensed MIT and the underlying library `rpi_ws218x` is licensed as BSD 2-Clause, which is more or less MIT.

@jibi - for the sake of consistency, could we please relicense Unicornd as MIT?

jibi, 2020-03-06:
@Gadgetoid yes, absolutely fine for me to relicense it :+1: (although it looks like there have been other contributors who worked on that)

Gadgetoid, 2020-03-12:
@jibi thank you! That should make things simpler.

closed 2020-03-16 by Gadgetoid