pimoroni/unicorn-hat #140: Doesn't work on raspberry pi 4

axelsimon, 2020-10-14:
Hi there,

A `pip3 install` of unicornhat works fine, but an installer using the installer at https://get.pimoroni.com/unicornhat warns that:
> This hardware is not supported, sorry!

Further attempting to `git clone` and run `detect.py` shows the following error:
> Note: Your Unicorn HAT must be plugged in before boot to detect properly!
>
> We couldn't find a connected EEPROM, so we'll assume you're using a Unicorn pHAT.

I'm really late to the party, and i have no real idea of whether `unicornhat` (which i realise has now been superseded by newer products) can be updated to work on the RPi4.

What would it take?

axelsimon, 2020-10-15:
Right, nothing works however, even if the install script completed correctly on the 32-bit version of raspios.

`detect.py` thinks a pHAT is connected when it's a HAT. None of the other examples do anything.

tschroeder13, 2020-10-18:
Hey all. 
Same problem for me.

KR

Gadgetoid, 2020-10-19:
I'll grab an older Unicorn HAT and look into this-

It's worth noting that `detect.py` will only ever find the EEPROM if the HAT is connected *before* booting the Pi, since the HAT EEPROM is only read by the system once at startup.

As far as I know, the rpi_ws281x library - which forms the core of Unicorn HAT and is what actually drives the pixels - still works.

JTucker48, 2020-10-19:
Hi all,
I'm also seeing the same issue with a raspberry 4b on raspiOS both with 32 and 64 bit. Below is the output I receive when running the detect.py script. As seen below, it does find something called "Unicorn HAT" but then doesnt seem to think its the right thing. I have also made sure the HAT is on properly and also checked that it does work fine with an older Pi (2b). Potentially it just thinks its a different product and this needs to be added to the library?
Any tips / pointing in the right direction would be super helpful and I'm happy to try stuff out if anyone has any ideas.

Many Thanks,
JT

> Unicorn HAT Detect-o-matic v2.0
> 
> Note: Your Unicorn HAT must be plugged in before boot to detect properly!
> 
> 
> Hold up, we found "Unicorn HAT" connected to your Pi!
> 
> Go walk the plank, landlubber! That ain't no Unicorn HAT
> 
> If this is wrong, chances are you've swapped HATs or pHATs without rebooting.
> 
> You should use the following in your code for Unicorn HAT:
> 
>     import unicornhat
>     unicornhat.set_layout(unicornhat.HAT)
> 
> Or for Unicorn pHAT:
> 
>     import unicornhat
>     unicornhat.set_layout(unicornhat.PHAT)

Gadgetoid, 2020-10-19:
The EEPROM is a bit of a red herring, since whether it detects or not has little to no bearing on whether the board is working, or will work.

I've just set up via the installer on my Pi 4 and run the examples, *and* `detect.py` with no issues followed by the examples which work fine on a Unicorn HAT board.

I suspect what's happening here is that you're all attempting to use Unicorn HAT HD boards with the Unicorn HAT software. If your board *does not* look like this:

[image]

Then you should be using this library here: https://github.com/pimoroni/unicorn-hat-hd

I'll see if I can add some helpful pointers to `detect.py` to mention the detected board is not supported, since confusing "Unicorn HAT" and "Unicorn HAT HD" software libraries is all too easy to do.

axelsimon, 2020-10-19:
Hi @Gadgetoid, thanks for chiming in. Good to hear this is working on your end and that there are no inherent issues with using the Unicorn HAT on a Raspberry Pi 4.

I can confirm i'm using the non-HD version, but i've now established that my rpi4 has hardware issues (the USB ports have stopped working in any useful way) and am getting a replacement. I'll try again with the new board, see if i run into the same problem.

Gadgetoid, 2020-10-19:
I've just noticed that the library determines "Unicorn HAT" to be a "Unicorn HAT" but the new `detect.py` in GitHub determines "unicornhat" to be a "Unicorn HAT." This discrepency might explain some of the issues with auto-detection.

If boot time EEPROM reading has worked, you should be able to:

```
cat /proc/device-tree/hat/product
```

It would be interesting to know what product name crops up there!

In all cases even an invalid detection should result in *something* happening on the Unicorn HAT, since the library will still output approximately the right signal. Driving a HAT like a pHAT will normally light just half of the HAT, for example.

It's also possible to modify the examples to `unicorn.set_layout(unicorn.HAT)`.

tschroeder13, 2020-10-19:
Hey all, thank you for your support.
I can report, that I got it working. I used [Pogo pins](https://shop.pimoroni.com/products/pogo-a-go-go-solderless-gpio-pogo-pins) and as Braden commented in the [reviews](https://shop.pimoroni.com/products/unicorn-phat#show-reviews) I also flipped the pins upside down and ... "let there be light"

One side info: I have no /proc/device-tree/hat although it is working now.

so I fixed it. 
Many thanks!

closed 2023-01-20 by Gadgetoid